mushorg/conpot #459: Modbus server "Illegal data address" exception

jnzhuang, 2019-07-07:
I tried to use conpot as a modbus slave, and I used libmodbus to generate a serial of modbus requests sent to conpot. Unfortunately, each the response packets contains an Exception of “Illegal data address”. (As an example, please take a look at the snapshot of attached pcap file, comprising only one request and one response in modbus protocol). I tried to look up the answer online, and found this:
https://github.com/mushorg/conpot/issues/288, which suggests modify the starting address of COILS to 0. But the solution does not work for me. So I am writing to ask if someone can give me some advice to solve the problem.
Thank you very much.
[image]

xandfury, 2019-07-08:
Could you share the details from the XML template? I am interested in looking at the slave id etc.

jnzhuang, 2019-07-08:
Thank you very much. Please refer to the attached file. Actually it is just the original default modbus.xml file with only one modification that I changed mode “serial” to “tcp”. My pcap file is also attached. 

I tried to read your code yesterday, and the code is written nicely. I found that the exception is raised by self._write_single_coil, which is a function in modbus-tk. Just for your reference. 

Thank you very much.

Best regards,
Jianing Zhuang

creolis, 2019-07-08:
Hey Jianing,

please do not send the attachment via eMail but post the content here on Github :)

jnzhuang, 2019-07-08:
Oh, oh. I am so sorry. I will post it on GitHub.

jnzhuang, 2019-07-08:
Thank you very much for your attention. Below is the xml file I used. I remember I also changed port number to 502.

===================== modbus.xml ==========================
<modbus enabled="True" host="0.0.0.0" port="502">
    <device_info>
        <VendorName>Siemens</VendorName>
        <ProductCode>SIMATIC</ProductCode>
        <MajorMinorRevision>S7-200</MajorMinorRevision>
    </device_info>
    <mode>tcp</mode>
    <delay>100</delay>
    <slaves>
        <slave id="0">
            <blocks>
                <block name="memoryModbusSlave0BlockA">
                    
                    <type>COILS</type>
                    <starting_address>0</starting_address>
                    <size>512</size>
                    <content>memoryModbusSlave0BlockA</content>
                </block>
                <block name="memoryModbusSlave0BlockB">
                    
                    <type>DISCRETE_INPUTS</type>
                    <starting_address>10001</starting_address>
                    <size>32</size>
                    <content>memoryModbusSlave0BlockB</content>
                </block>
            </blocks>
        </slave>
        <slave id="255">
            <blocks>
                <block name="memoryModbusSlave255BlockA">
                    
                    <type>COILS</type>
                    <starting_address>0</starting_address>
                    <size>512</size>
                    <content>memoryModbusSlave255BlockA</content>
                </block>
                <block name="memoryModbusSlave255BlockB">
                    
                    <type>DISCRETE_INPUTS</type>
                    <starting_address>10001</starting_address>
                    <size>32</size>
                    <content>memoryModbusSlave255BlockB</content>
                </block>
            </blocks>
        </slave>
        <slave id="1">
            <blocks>
                <block name="memoryModbusSlave1BlockA">
                    
                    <type>COILS</type>
                    <starting_address>0</starting_address>
                    <size>512</size>
                    <content>memoryModbusSlave1BlockA</content>
                </block>
                <block name="memoryModbusSlave1BlockB">
                    
                    <type>DISCRETE_INPUTS</type>
                    <starting_address>10001</starting_address>
                    <size>32</size>
                    <content>memoryModbusSlave1BlockB</content>
                </block>
            </blocks>
        </slave>
        <slave id="2">
            
            <blocks>
                <block name="memoryModbusSlave2BlockC">
                    
                    <type>ANALOG_INPUTS</type>
                    <starting_address>30001</starting_address>
                    <size>8</size>
                    <content>memoryModbusSlave2BlockC</content>
                </block>
                <block name="memoryModbusSlave2BlockD">
                    
                    <type>HOLDING_REGISTERS</type>
                    <starting_address>40001</starting_address>
                    <size>8</size>
                    <content>memoryModbusSlave2BlockD</content>
                </block>
            </blocks>
        </slave>
    </slaves>
</modbus>

jnzhuang, 2019-07-08:
Thank you very much for your attention. The file modbus.xml is attached. I remember also that I changed the port number to 502.
[modbus.txt](https://github.com/mushorg/conpot/files/3369067/modbus.txt)

jnzhuang, 2019-07-08:
In my request_pdu, the data_address field parsed by _write_single_coil is 304. How could I modify xml file accordingly? @xandfury @creolis

jnzhuang, 2019-07-08:
Oh, I found it. 
In conpot/templates/default/template.xml
Just to change range(0,128) to range(0,x) where x is something larger than 304 will be fine.
I focused too much in modbus.xml.

closed 2019-07-08 by jnzhuang