mushorg/conpot #461: error when execute docker run

tXambe, 2019-07-26:
Hello,

When execute:
 
docker run -it -p 80:8800 -p 102:10201 -p 502:5020 -p 161:16100/udp -p 47808:47808/udp -p 623:6230/udp -p 21:2121 -p 69:6969/udp -p 44818:44818 --network=bridge conpot

I got this error:

docker: Error response from daemon: pull access denied for conpot, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.

Can anyone help me ?

A greeting and thanks

xandfury, 2019-07-28:
You need to pull the image before running it. Try:
https://github.com/mushorg/conpot#via-a-pre-built-image

tXambe, 2019-07-29:
Hello,

I follow this steps but I got other error:

    Install Docker
    Clone this repo with git clone https://github.com/mushorg/conpot.git and cd conpot/docker
    Run docker build -t conpot . ( when finish this command )
    
   -- Docs: https://docs.pytest.org/en/latest/warnings.html
    ======== 6 failed, 156 passed, 5 skipped, 11 warnings in 179.92 seconds ========
   The command '/bin/sh -c py.test -v' returned a non-zero code: 1

    Run docker run -it -p 80:8800 -p 102:10201 -p 502:5020 -p 161:16100/udp -p 47808:47808/udp -p 623:6230/udp -p 21:2121 -p 69:6969/udp -p 44818:44818 --network=bridge conpot --> If run this command:

sudo docker run -it -p 80:8800 -p 102:10201 -p 502:5020 -p 161:16100/udp -p 47808:47808/udp -p 623:6230/udp -p 21:2121 -p 69:6969/udp -p 44818:44818 --network=bridge conpot
Unable to find image 'conpot:latest' locally
docker: Error response from daemon: pull access denied for conpot, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.

closed 2019-07-29 by tXambe

tXambe, 2019-07-31:
Hello,
I follow the steps of link, but don't run.

sudo docker pull honeynet/conpot
[sudo] password for admlc:
Using default tag: latest
latest: Pulling from honeynet/conpot
Digest:
sha256:cd93e88d9e44b020db691fc4c75cb29e76b5e90ddbc408aca26e6c78c5646976
Status: Image is up to date for honeynet/conpot:latest
docker.io/honeynet/conpot:latest

When I run " sudo docker run -it -p 80:80 -p 102:102 -p 502:502 -p
161:161/udp --network=bridge honeynet/conpot:latest /bin/sh "
I only get a prompt " ~ $".

If I execut  "conpot -f --template default" I have this error:

~ $ conpot -f --template default
/bin/sh: conpot: not found
~ $

If I exit of this prompt, and execute " conpot -f --template default" I
have this error:

/opt/conpot/docker$ sudo conpot -f --template default
Traceback (most recent call last):
  File "/usr/local/bin/conpot", line 4, in <module>
    __import__('pkg_resources').run_script('conpot==0.6.0', 'conpot')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line
666, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line
1446, in run_script
    exec(code, namespace, namespace)
  File
"/usr/local/lib/python2.7/dist-packages/conpot-0.6.0-py2.7.egg/EGG-INFO/scripts/conpot",
line 28, in <module>
    from configparser import ConfigParser, NoSectionError, NoOptionError
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line
96, in __import__
    result = _import(*args, **kwargs)
  File
"/usr/local/lib/python2.7/dist-packages/configparser-3.7.4-py2.7.egg/configparser.py",
line 11, in <module>
    from backports.configparser import (
ImportError: cannot import name ConverterMapping

A greeting and thanks very much for your support.

xandfury, 2019-07-31:
>  /usr/local/lib/python2.7/dist-packag
You need to run `conpot` with python3 :slightly_smiling_face: 

You should try installing using this guide: https://conpot.readthedocs.io/en/latest/installation/install.html

We are in the process of updating the docs. Would love some feedback :wink:

tXambe, 2019-08-06:
Hello,

I run fresh installation of conpot with pyton3 but when I run conpot I have this error.

Invalid arguments supplied. Please check that you pass both template and config arguments before running Conpot

A greeting