mushorg/conpot #468: Conpot keeps restarting

Norup123, 2019-10-09:
It worked for a while, and then all of the sudden it just started breaking constantly
conpot.log looks like this:
[image]
[image]

its been deployed with MHN

creolis, 2019-10-09:
Hey! Is there any way to get a better view at your logs?

First of all, screenshots are really inconvenient since they are an accessibility nightmare and they can't be indexed and searched -> but more importantly in your case is that all important messages are truncated, so we have virtually no information to start with..

Norup123, 2019-10-09:
``
[conpot.log](https://github.com/mushorg/conpot/files/3706294/conpot.log)

creolis, 2019-10-09:
Oh it's fine, now we can actually read what's going on :) 

`2019-10-09 10:11:51,828 Stopping because <Greenlet at 0x7fa02f969890: <bound method HTTPServer.start of <conpot.protocols.http.web_server.HTTPServer object at 0x7fa02f92e110>>('0.0.0.0', 80)> died: [Errno 98] Address already in use`

Did you try to resolve this?

Norup123, 2019-10-09:
Not sure what it means. Is port 80 busy with something else?

creolis, 2019-10-09:
> ahh shit i think maybe my Dionaea was also using port 80

;-) Well, that was an easy one.
btw.: if you want to find out which process grabbed port 80,
try 

`sudo netstat -anp | grep LISTEN | grep -v LISTENING | grep 80 | awk '{print $7}'`

(root permissions are used to resolve the PID)

closed 2019-10-09 by creolis