mushorg/conpot #469: fs.errors.ResourceReadOnly: resource 'ftp' is read only

Mr-later, 2019-10-30:
please help me ,Here is the error message and configuration：

~/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/bin $ python3 conpot -c ../testing1.cfg --template
 default
WARNING:scapy.runtime:No route found for IPv6 destination :: (no default route?)

                       _
   ___ ___ ___ ___ ___| |_
  |  _| . |   | . | . |  _|
  |___|___|_|_|  _|___|_|
              |_|

  Version 0.6.0
  MushMush Foundation

args.force================ False
mei shiyong testing.cfg===========
2019-10-30 08:50:58,264 Config file found!
2019-10-30 08:50:58,277 Starting Conpot using template: /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default
2019-10-30 08:50:58,277 Starting Conpot using configuration found in: /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/../testing1.cfg
fs_url= /tmp/
data_fs_url= tar:///home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/data.tar
args.temp_dir= ConpotTempFS False
conpot.__path__= ['/home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot']
conpot.__path__[0]= /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot
2019-10-30 08:50:58,298 Can't find the temp directory. Conpot VFS would be kept at : /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/ConpotTempFS
2019-10-30 08:50:58,298 Serving /tmp/ as file system. File uploads will be kept at : tar:///home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/data.tar
2019-10-30 08:50:58,299 Opening path tar:///home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/data.tar for persistent storage of files.
2019-10-30 08:50:58,320 Initializing Virtual File System at /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/ConpotTempFS/__conpot__f8pwxo7w. Source specified : /tmp/
 Please wait while the system copies all specified files
2019-10-30 08:51:03,168 Fetched 36.110.67.38 as external ip.
protocol_name= modbus
protocol_template= /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default/modbus/modbus.xml
root_template_directory= /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default
args= Namespace(config='/home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/../testing1.cfg', force=False, logfile='conpot.log', mibpaths=[], raw_mib=[], temp_dir='ConpotTempFS', template='default', verbose=False)
2019-10-30 08:51:03,218 Conpot modbus initialized
2019-10-30 08:51:03,250 Found and enabled modbus protocol.
protocol_name= s7comm
protocol_template= /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default/s7comm/s7comm.xml
root_template_directory= /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default
args= Namespace(config='/home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/../testing1.cfg', force=False, logfile='conpot.log', mibpaths=[], raw_mib=[], temp_dir='ConpotTempFS', template='default', verbose=False)
2019-10-30 08:51:03,258 Conpot S7Comm initialized
2019-10-30 08:51:03,259 Found and enabled s7comm protocol.
protocol_name= kamstrup_meter
protocol_template= /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default/kamstrup_meter/kamstrup_meter.xml
root_template_directory= /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default
args= Namespace(config='/home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/../testing1.cfg', force=False, logfile='conpot.log', mibpaths=[], raw_mib=[], temp_dir='ConpotTempFS', template='default', verbose=False)
protocol_name= kamstrup_management
protocol_template= /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default/kamstrup_management/kamstrup_management.xml
root_template_directory= /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default
args= Namespace(config='/home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/../testing1.cfg', force=False, logfile='conpot.log', mibpaths=[], raw_mib=[], temp_dir='ConpotTempFS', template='default', verbose=False)
protocol_name= http
protocol_template= /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default/http/http.xml
root_template_directory= /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default
args= Namespace(config='/home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/../testing1.cfg', force=False, logfile='conpot.log', mibpaths=[], raw_mib=[], temp_dir='ConpotTempFS', template='default', verbose=False)
2019-10-30 08:51:03,266 Found and enabled http protocol.
protocol_name= snmp
protocol_template= /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default/snmp/snmp.xml
root_template_directory= /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default
args= Namespace(config='/home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/../testing1.cfg', force=False, logfile='conpot.log', mibpaths=[], raw_mib=[], temp_dir='ConpotTempFS', template='default', verbose=False)
2019-10-30 08:51:03,284 Found and enabled snmp protocol.
protocol_name= bacnet
protocol_template= /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default/bacnet/bacnet.xml
root_template_directory= /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default
args= Namespace(config='/home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/../testing1.cfg', force=False, logfile='conpot.log', mibpaths=[], raw_mib=[], temp_dir='ConpotTempFS', template='default', verbose=False)
2019-10-30 08:51:03,300 Conpot Bacnet initialized using the /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default/bacnet/bacnet.xml template.
2019-10-30 08:51:03,301 Found and enabled bacnet protocol.
protocol_name= ipmi
protocol_template= /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default/ipmi/ipmi.xml
root_template_directory= /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default
[... 31 lines cut ...]
path= ftp
Traceback (most recent call last):
  File "conpot", line 476, in <module>
    main()
  File "conpot", line 393, in main
    server = server_class(protocol_template, root_template_directory, args)
  File "/home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/core/protocol_wrapper.py", line 27, in __init__
    self.wrapped = cls(*args, **kwargs)
  File "/home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/protocols/ftp/ftp_server.py", line 225, in __init__
    self.handler.config = FTPConfig(self.template)
  File "/home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/protocols/ftp/ftp_server.py", line 103, in __init__
    self._init_fs()        # Initialize FTP file system.
  File "/home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/protocols/ftp/ftp_server.py", line 117, in _init_fs
    perms=self.default_perms)
  File "/home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/core/__init__.py", line 55, in add_protocol
    return virtualFS.add_protocol(protocol_name, data_fs_subdir, vfs_dst_path, src_path, owner_uid, group_gid, perms)
  File "/home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/core/virtual_fs.py", line 121, in add_protocol
    sub_data_fs = self.data_fs.makedir(path=data_fs_subdir)
  File "/home/conpot/.local/lib/python3.6/site-packages/fs/tarfs.py", line 400, in makedir
    raise errors.ResourceReadOnly(path)
fs.errors.ResourceReadOnly: resource 'ftp' is read only

[image]
vi testing1.cfg
[image]

Mr-later, 2019-10-31:
I find this ，Is MySQL not supported?
[image]

Mr-later, 2019-11-01:
If you want to save to MySQL
You should modify this file：mysql_log.py,use pymysql  Because Python 3 doesn't support MySqldb
Saving to the database is not recommended because it may cause database information disclosure

------
Now I have a new question，How can I customize the log saving address according to the format of one file a day。The source code is like this, how should I change it。like /var/log/{date}.json
[image]

thank you

Mr-later, 2019-11-02:
I fix it myself .thank you

closed 2019-11-02 by Mr-later

creolis, 2019-11-04:
[image]