mushorg/conpot #478: does not work! HELP PLZ. ubuntu1~18.04.1

cr3wcu7, 2020-01-22:
Installed docker at https://docs.docker.com/install/linux/docker-ce/ubuntu/

Run: `docker pull honeynet/conpot` -OK

1) Run: `docker run -it -p 80:80 -p 102:102 -p 502:502 -p 161:161/udp --network=bridge honeynet/conpot:latest /bin/sh`

2) `~ $ conpot -f --template default`
/bin/sh: conpot: not found

prints no found....ok 

1.1) Run: `docker run -it -p 80:80 -p 102:102 -p 502:502 -p 161:161/udp --network=bridge honeynet/conpot`

Prints:
```
root@r00t-VirtualBox:~# docker run -it -p 80:80 -p 102:102 -p 502:502 -p 161:161/udp --network=bridge honeynet/conpot
WARNING:scapy.runtime:No route found for IPv6 destination :: (no default route?)

                       _
   ___ ___ ___ ___ ___| |_
  |  _| . |   | . | . |  _|
  |___|___|_|_|  _|___|_|
              |_|

  Version 0.6.0
  MushMush Foundation

WARNING:root:--force option specified. Using testing configuration
2020-01-22 15:59:21,565 --force option specified. Using testing configuration
2020-01-22 15:59:21,568 Starting Conpot using template: /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default
2020-01-22 15:59:21,568 Starting Conpot using configuration found in: /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/testing.cfg
WARNING:conpot.core.virtual_fs:Using default FS path. tar:///home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/data.tar
2020-01-22 15:59:21,578 Using default FS path. tar:///home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/data.tar
2020-01-22 15:59:21,580 Initializing Virtual File System at /tmp/__conpot__mwyfzmi1. Source specified : tar:///home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/data.tar
 Please wait while the system copies all specified files
2020-01-22 15:59:21,833 Fetched 92.49.180.58 as external ip.
2020-01-22 15:59:21,839 Conpot modbus initialized
2020-01-22 15:59:21,839 Found and enabled modbus protocol.
2020-01-22 15:59:21,842 Conpot S7Comm initialized
2020-01-22 15:59:21,842 Found and enabled s7comm protocol.
2020-01-22 15:59:21,844 Found and enabled http protocol.
2020-01-22 15:59:21,845 Found and enabled snmp protocol.
2020-01-22 15:59:21,847 Conpot Bacnet initialized using the /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default/bacnet/bacnet.xml template.
2020-01-22 15:59:21,847 Found and enabled bacnet protocol.
2020-01-22 15:59:21,849 IPMI BMC initialized.
2020-01-22 15:59:21,850 Conpot IPMI initialized using /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default/ipmi/ipmi.xml template
2020-01-22 15:59:21,850 Found and enabled ipmi protocol.
2020-01-22 15:59:21,853 Class    22/0x0016, Instance   1, Attribute     1 <== [{'class': 22}, {'instance': 1}, {'attribute': 1}]
2020-01-22 15:59:21,854 Class    22/0x0016, Instance   1, Attribute     2 <== [{'class': 22}, {'instance': 1}, {'attribute': 2}]
2020-01-22 15:59:21,854 Class    22/0x0016, Instance   1, Attribute     1 <== [{'class': 22}, {'instance': 1}, {'attribute': 1}]
2020-01-22 15:59:21,856 Class    22/0x0016, Instance   1, Attribute     3 <== [{'class': 22}, {'instance': 1}, {'attribute': 3}]
2020-01-22 15:59:21,856 Class    22/0x0016, Instance   1, Attribute     1 <== [{'class': 22}, {'instance': 1}, {'attribute': 1}]
2020-01-22 15:59:21,856 Class    22/0x0016, Instance   1, Attribute     2 <== [{'class': 22}, {'instance': 1}, {'attribute': 2}]
2020-01-22 15:59:21,857 Found and enabled enip protocol.
2020-01-22 15:59:21,861 Creating persistent data store for protocol: ftp
2020-01-22 15:59:21,876 FTP Serving File System at /data/ftp/ in vfs. FTP data_fs sub directory: /ftp
2020-01-22 15:59:21,897 Found and enabled ftp protocol.
2020-01-22 15:59:21,898 Creating persistent data store for protocol: tftp
2020-01-22 15:59:21,912 TFTP Serving File System at /data/tftp/ in vfs. TFTP data_fs sub directory: /tftp
2020-01-22 15:59:21,918 Found and enabled tftp protocol.
2020-01-22 15:59:21,919 No proxy template found. Service will remain unconfigured/stopped.
2020-01-22 15:59:21,919 Modbus server started on: ('0.0.0.0', 5020)
2020-01-22 15:59:21,922 S7Comm server started on: ('0.0.0.0', 10201)
2020-01-22 15:59:21,922 HTTP server started on: ('0.0.0.0', 8800)
2020-01-22 15:59:22,539 SNMP server started on: ('0.0.0.0', 16100)
2020-01-22 15:59:22,542 Bacnet server started on: ('0.0.0.0', 47808)
2020-01-22 15:59:22,543 IPMI server started on: ('0.0.0.0', 6230)
2020-01-22 15:59:22,543 handle server PID [    1] running on ('0.0.0.0', 44818)
2020-01-22 15:59:22,544 handle server PID [    1] responding to external done/disable signal in object 140423589885896
2020-01-22 15:59:22,545 FTP server started on: ('0.0.0.0', 2121)
2020-01-22 15:59:22,545 Starting TFTP server at ('0.0.0.0', 6969)
```
http://localhost -  **not available**
There is no reaction when contacting localhost.

What's my mistake ?
HELLLP PLIZ =)

Finally run conpot -f --template default

creolis, 2020-01-22:
> 2020-01-22 15:59:21,922 HTTP server started on: ('0.0.0.0', 8800)

... please mind the default HTTP listening port, according to the log output you attached

closed 2020-01-22 by creolis

cr3wcu7, 2020-01-23:
I paid attention to it, the result is the same, im used.
I used the following:  sudo docker run -it -p 80:8800 -p 102:10201 -p
502:5020 -p 161:16100/udp --network=bridge honeynet/conpot
Also did not give results.

Help me pliz

xandfury, 2020-01-24:
@cr3wcu7 Can you share the output of `sudo docker ps`? Also `sudo netstat -antp | grep 80`