mushorg/conpot #480: Logging to MySQL, not working

KKAB-FK, 2020-02-20:
I have a Conpot fully setup to look as close to a real PLC as possible ie. nearly finished :)
According to a previous statement https://github.com/mushorg/conpot/issues/469#issuecomment-548200494 the logging to a MySQL database is no longer supported?

What is the solution to this?, What do i need to do in order to be able to log to a MySQL database?

BR.
Felix

glaslos, 2020-11-23:
Hi Felix,
Did you find a way around it? MySQL was discontinued during the python3 migration and we are about to completely remove it :) If you really need MySQL support, I'd suggest using the event stream and writing your own MySQL writer.

KKAB-FK, 2020-12-22:
Hi Lukas,
Sorry for the late response!

Yeah i saw that to :(
I don't necessarily need MySQL support, a Syslog would work just as good hopefully, unfortunately i haven't got that to work either :(, hence why i opted for MySQL.
I'll give the Syslog a work once more ;). Btw. is it verified that Syslog will work?

Seasons Greetings from Sweden :)
//Felix

KKAB-FK, 2023-02-07:
Fixed:

Remove the this whole line:  "" ; udp (sends to host:port), dev (sends to device) "" at the Syslog section in your .cfg file.
And only write "socket = udp" in order for syslog to work :)
Simple mistake by me ;)

closed 2023-02-07 by KKAB-FK