mushorg/conpot #482: bug in IEC104_server.py

Noggerman, 2020-04-01:
Hey, I found a bug in [IEC104_server.py](https://github.com/mushorg/conpot/blob/master/conpot/protocols/IEC104/IEC104_server.py).
When confronted with an input of only one byte, the server gets stuck in the while statement at line 58.
```
while request and len(request) < 2:
	new_byte = sock.recv(1)
	request += new_byte
```	
The permanent loop results in a cpu overload and it could be exploited for denial-of-service attacks.

The same bug occurs, when the server is fed an input which is one byte longer than the given length of APDU, for example 68040100000000 and you try to send another command.

 A possible fix would be to change the while statement at line 58 to an if statement:
```
if request and len(request) < 2:
	new_byte = sock.recv(1)
	request += new_byte
```	
 If the loop was intended, you could add a counter or a timer to limit the duration of the loop:
```
for i in range(50):
	if request and len(request) < 2:
		new_byte = sock.recv(1)
		request += new_byte
	else:
		break
```

xandfury, 2020-04-11:
@Noggerman Which template are you running? Check here:
https://github.com/mushorg/conpot/blob/b2836c848f6998eaf41597931116559f96350456/conpot/protocols/IEC104/IEC104_server.py#L48-L49

A timeout should be raised if the key `T_3` is set on the template.

Noggerman, 2020-04-14:
I'm running the template "IEC104" with the testing configuration (`conpot -f -t IEC104`). A timeout was never raised. The key is being set in line 363 of [template.xml](https://github.com/mushorg/conpot/blob/master/conpot/templates/IEC104/template.xml). It should trigger after 20 seconds, but it doesn't.

xandfury, 2020-04-14:
Thanks for the information. In that case, it needs to be investigated :slightly_smiling_face:

glaslos, 2026-09-14:
Addressed with 040fcf4e0bea0292e3003fd29775aa9efa502b3e

closed 2026-09-14 by glaslos