mushorg/conpot #494: Code style is not enforced

srenfo, 2020-05-12:
The code has plenty of PEP 8 violations. That's not surprising given its size and its many authors.

For example, _even ignoring line length_, `pycodestyle` finds 300 violations on the current `master`:

```shell
$ pycodestyle bin/ conpot/ setup.py --ignore=E501 | wc
    300    2175   25226
```

There are plenty of tools to combat this, including `pycodestyle`/`flake8`, `pylint`,  or `black`.

**Suggestion:** pick one and include it in the test suite.

The biggest issues is that the first changeset would be large.

I have some experience with those tools and can weigh in if desired.

xandfury, 2020-05-13:
I really like black. Although we can just integrate something like codacy into our pipeline. That would also cover static analysis.

glaslos, 2020-07-05:
Created a GitHub Action with black, let me know what you think.

glaslos, 2020-07-05:
I run `black` fixing the styling issues. PR is here #501

srenfo, 2020-07-17:
> Created a GitHub Action with black, let me know what you think.

That's really cool.

You could add `black --check conpot/ bin/` to `tox.ini` as well for local testing, but that would do the checks twice. I don't know if it's worthwhile. :shrug:

closed 2020-07-17 by srenfo