mushorg/conpot #503: Missing documentation for local execution of pytest/tox

srenfo, 2020-07-20:
There is no explicit documentation on how to run the test suite.

As for requirements, at the very least `ipmitool` needs to be installed on Debian/derivatives, in addition to the other requirements. Currently, that fact is only implicitly documented in `travis.yml` and `Dockerfile`. `ipmitool` is not required at runtime and is therefore (appropriately?) missing from [install.rst](https://github.com/mushorg/conpot/blob/master/docs/source/installation/install.rst). Also, `tox` should be mentioned somewhere, since right now only the existence of `tox.ini` tips you off that that's how the tests are run.

Suggestions:
1. add a subsection to the install documentation since that's where the other requirements are documented
1. add a small section to the top-level README.md
1. roll it into a CONTRIBUTING.md (since that's where development-oriented documentation could live)

I can try my hand at a PR. I would lean towards adding it to the README.

glaslos, 2020-07-21:
How about extending https://conpot.readthedocs.io/en/latest/development/guidelines.html ?
Maybe add more deep links to the README?

glaslos, 2026-09-14:
Fixed in f8abc293898a2ad2ad8685cb5e3d1c2d157f9c08

closed 2026-09-14 by glaslos