mushorg/conpot #504: Conpot K8S

Xsenonn, 2020-07-21:
## Expected Behavior

Kubernetes should start a few pods containing Conpot instance. 
Conpot should start and put logs on mounted volume.
## Current Behavior

Kubernetes creates Deployment, but Conpot Pods crash giving error below.
[image]
All volumes for logs are mounting properly and conpot has write access to them. 
Custom Image of conpot has been built based on Dockerfile prepared by T-Mobile for tpotce Honeypot Bundle. Upon building image I've bumped on error like this:
`Cannot uninstall 'distlib'. It is a distutils installed project and             
thus we cannot accurately determine which files belong to it which 
would lead to only a partial uninstall.`
The error above disappears when pip in Dockerfile is invoked with `--ignore-installed` option.

## Steps to Reproduce

1. Install Minikube 
1a. Deploy MetalLB on Minikube
1b. Configure IP address range for MetalLB to hand out
2. Deploy Conpot using link below (Conpot image has been built based on T-Mobile tpotce conpot Dockerfile)
https://gist.github.com/Xsenonn/874409d0e94fa249d73cd454c21824a4
3. Deploy conpot-service using link below
https://gist.github.com/Xsenonn/0d57d96ed4ed9d3efac2f57c2229c93c

glaslos, 2020-10-27:
It seems to be a permissions issue as you can see on line 5 of the screenshot. I'm closing the issue as it should be filed with T-Pot, cc @t3chn0m4g3

closed 2020-10-27 by glaslos

t3chn0m4g3, 2020-10-28:
@glaslos Works perfectly fine in T-Pot 😃 Seems this is / was a custom build.