mushorg/conpot #507: Error in log and snap7 can't receive avaliable messages from conpot

lemorn, 2020-08-13:
Hello! I met a problem when I ran conpot in default template. When I run conpot on my cloud server(OS is Ubuntu and the conpoot version is 0.6.0), a error appeared:
[image]
and my snap7 can't receive avaliable messages about this "S7-PLC".
[image]
How can I solve these?

I also want to know if i can deploy a new honeypot (like mitsubish IQ-R PLC) by conpot? Which part of the code should I modify to deploy a new honeypot?If I want to modify some messgaes sent by conpot, such as type of cpu, what should I do?

And there seems to be a small problem. When running the docker container, there seems to be some problems with the port mapping. After running the container, it shows that these ports are opened. But as far as I know, the ports corresponding to s7, modbus, and http should be 102, 502, and 80. Maybe this has any special purpose?
[image]

Thank you very much!

glaslos, 2020-10-27:
1) what requests are you sending causing the error?
2) have a look into the conpot/templates folder. It should give you an idea how to emulate devices
3) check the conpot/templates folder and the files inside. You can also use @docker-compose@ to run the container.

Han-zb, 2020-11-04:
1. "snap7 can't receive avaliable messages", this is probably because the conpot is emulating a s7-200 PLC,  different series of plc needs different  connection codes when sending s7comm, tpkt and cotp, you can have a try using the codes below:
```
import socket

connection= ("ip", port)

sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
sock.connect(connection)

#Connect plc
cotp_conn_str = "0300001611e00000000600c1020100c2020102c0010a"
cotp_conn = bytes.fromhex(cotp_conn_str)
sock.sendall(cotp_conn)
data = sock.recv(1024)
print("response to cotp: ", data)

s7_conn_str = "0300001902f08032010000ccc100080000f0000001000103c0"
s7_conn = bytes.fromhex(s7_conn_str)
sock.sendall(s7_conn)
data2 = sock.recv(1024)
print("response to s7_conn: ", data2)

#Send requests
req_str = "0300002102f080320700001200000800080001120411440100ff090004001c0000"#unit info
req = bytes.fromhex(req_str)
sock.sendall(req)
data3 = sock.recv(1024)
print("response to request: ", data3)
```

2. the conpot is running in the docker, the ports 5020, 10201, 8800 are the docker container ports. And when you first create a container to run conpot using command
`docker run -it -p 80:8800 -p 102:10201 -p 502:5020 -p 161:16100/udp --network=bridge honeynet/conpot:latest /bin/sh` 
actually there is a port mapping, you can modified them in the templates/default/***/***.xml

soso288, 2021-04-17:
@Han-zb I don't really understand where to put your code. I tried putting it in init.py but I got a connection refused error.

Also can you tell me how to correctly modify the s7comm.xml file, so that I get similar output when usind plcscan?

223.171.35.149:102 S7comm (src_tsap=0x100, dst_tsap=0x102)
  Module                   : 6ES7 312-1AE14-0AB0  v.0.7    	(36455337203331322d31414531342d304142302000c000070001)
  Basic Hardware           : 6ES7 312-1AE14-0AB0  v.0.7    	(36455337203331322d31414531342d304142302000c000070001)
  Basic Firmware           :                      v.3.3.12 	(202020202020202020202020202020202020202000c05603030c)
  Unknown (129)            : Boot Loader           A%      	(426f6f74204c6f61646572202020202020202020000041250c0c)

I tried editing the file with the information above but I get this error in plcscan:

Scan start...                          
127.0.0.1:10201 [Errno 104] Connection reset by peer
127.0.0.1:10201 S7comm (src_tsap=0x100, dst_tsap=0x102)
  [ERROR][S7Protocol] Unknown TPKT format
  [ERROR][S7Protocol] Unknown TPKT format
Scan complete 

and this in conpot:

ERROR:conpot.protocols.s7comm.s7_server:Exception caught , remote: 127.0.0.1. (775b5d19-d071-48ab-8e1d-d8504e9b7f41)
Traceback (most recent call last):
  File "/home/soso/conpot/lib/python3.6/site-packages/conpot/protocols/s7comm/s7_server.py", line 170, in handle
    response_param, response_data = S7_packet.handle()
  File "/home/soso/conpot/lib/python3.6/site-packages/conpot/protocols/s7comm/s7.py", line 62, in handle
    return self.param_mapping[self.param][1]()
  File "/home/soso/conpot/lib/python3.6/site-packages/conpot/protocols/s7comm/s7.py", line 157, in request_diagnostics
    description, params, data = m(data_ssl_index)
  File "/home/soso/conpot/lib/python3.6/site-packages/conpot/protocols/s7comm/s7.py", line 181, in request_ssl_17
    str_to_bytes(self.data_bus.get_value(current_ssl['W#16#0001'])),
  File "/home/soso/conpot/lib/python3.6/site-packages/conpot/core/databus.py", line 43, in get_value
    assert key in self._data
AssertionError

glaslos, 2026-09-14:
Closed with b72bdc1df76ec397271c813f5c1c0efca4437c3a

closed 2026-09-14 by glaslos