mushorg/conpot #508: BacnetServer breaks with error

t3chn0m4g3, 2020-08-13:
I am getting the following error when trying to start the `default` template:
```
conpot_default         | Traceback (most recent call last):
conpot_default         |   File "src/gevent/greenlet.py", line 854, in gevent._gevent_cgreenlet.Greenlet.run
conpot_default         |   File "/usr/lib/python3.8/site-packages/conpot/protocols/bacnet/bacnet_server.py", line 108, in start
conpot_default         |     self.bacnet_app = BACnetApp(self.thisDevice, self.server)
conpot_default         |   File "/usr/lib/python3.8/site-packages/conpot/protocols/bacnet/bacnet_app.py", line 69, in __init__
conpot_default         |     super().__init__(self.localDevice, datagram_server)
conpot_default         |   File "/usr/lib/python3.8/site-packages/bacpypes/app.py", line 473, in __init__
conpot_default         |     ApplicationIOController.__init__(self, localDevice, localAddress, deviceInfoCache, aseID=aseID)
conpot_default         |   File "/usr/lib/python3.8/site-packages/bacpypes/app.py", line 397, in __init__
conpot_default         |     Application.__init__(self, *args, **kwargs)
conpot_default         |   File "/usr/lib/python3.8/site-packages/bacpypes/app.py", line 234, in __init__
conpot_default         |     self.localAddress = Address(localAddress)
conpot_default         |   File "/usr/lib/python3.8/site-packages/bacpypes/pdu.py", line 52, in __init__
conpot_default         |     self.decode_address(args[0])
conpot_default         |   File "/usr/lib/python3.8/site-packages/bacpypes/pdu.py", line 311, in decode_address
conpot_default         |     raise TypeError("integer, string or tuple required")
conpot_default         | TypeError: integer, string or tuple required
conpot_default         | 2020-08-13T15:54:43Z <Greenlet at 0x7fc1622fbae0: <bound method BacnetServer.start of <conpot.protocols.bacnet.bacnet_server.BacnetServer object at 0x7fc1620ae2e0>>('0.0.0.0', 47808)> failed with TypeError
```

Everything works fine up to and including commit 7a77329cd99cee9c37ee20e2f05a48952d8eece9.

glaslos, 2020-10-27:
@t3chn0m4g3 can you try the latest master?

t3chn0m4g3, 2020-10-28:
@glaslos Thank you, works perfectly fine now 😄

closed 2020-10-28 by t3chn0m4g3