mushorg/conpot #509: OID missing

t3chn0m4g3, 2020-08-13:
For IEC104 I am getting the following warning:
```
WARNING:conpot.protocols.snmp.command_responder:Skipped: OID for symbol ifOutUNcastPkts not found in MIB IF-MIB
```
I am using the following MIB database:
```
 wget https://github.com/wireshark/wireshark/raw/master/manuf -o /usr/share/wireshark/manuf
```
Do you have any suggestions?

srenfo, 2020-08-28:
You're seeing this because this [used to be a debug-level message rather than a warning](https://github.com/mushorg/conpot/commit/c0319e98e3fa8b6ce90ed9efc63de78ee2162cf8#diff-a4bb36aa9be0fe1cd8e4ef87403a5861L142-R141) until recently.

The fix is to remove `ifOutUNcastPkts` from the template, but read on if you want to keep it.

When @pudii added the IEC104 template, their IF-MIB (or maybe the device they modelled the template after) must have included `ifOutUNcastPkts`. However, the common [IF-MIB definition](https://github.com/net-snmp/net-snmp/blob/master/mibs/IF-MIB.txt) does not. But the latter is the IF-MIB that's likely installed on your system or that Conpot has downloaded on-demand.

So, to keep `ifOutUNcastPkts` you need to find an IF-MIB definition that contains `ifOutUNcastPkts`.

Put differently, since (what appears to be) the canonical IF-MIB does not include `ifOutUNcastPkts` I would classify the warning as a bug and remove the OID from all Conpot templates.

soso288, 2021-05-06:
@t3chn0m4g3 how exactly did you integrate the MIB into conpot? I tried, but I wasn't able to do it at all.

t3chn0m4g3, 2021-05-06:
@soso288 I did it, so you do not have to. You can see this in the docker folder for [T-Pot](https://github.com/telekom-security/tpotce).

soso288, 2021-05-06:
Thanks for your fast answer. I think I did it the same way (1. download IF-MIB 2. extend the template 3. start conpot with -m "MIB-directory" parameter) but the MIBBrowser I'm using is only showing the default MIB and "snmpInPkts, snmpOutPkts, snmpInBadVersions..." with value=0 after that. What could be the problem?

t3chn0m4g3, 2021-05-06:
The reason was given in the comment of @srenfo.

soso288, 2021-05-07:
Thank you I was able to fix it

closed 2026-09-14 by glaslos