mushorg/conpot #511: Error in modbus server

tomac, 2020-08-26:
```
Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 766, in gevent._greenlet.Greenlet.run
  File "/usr/lib/python3.6/site-packages/gevent/baseserver.py", line 26, in _handle_and_close_when_done
    return handle(*args_tuple)
  File "/usr/lib/python3.6/site-packages/conpot/protocols/modbus/modbus_server.py", line 125, in handle
    query, request, self.mode
  File "/usr/lib/python3.6/site-packages/conpot/protocols/modbus/slave_db.py", line 77, in handle_request
    request_pdu, broadcast=True)
  File "/usr/lib/python3.6/site-packages/conpot/protocols/modbus/slave.py", line 87, in handle_request
    (self.function_code, ) = struct.unpack(">B", request_pdu[:1])
struct.error: unpack requires a buffer of 1 bytes
2020-08-10T09:31:42Z <Greenlet at 0x7f2c8e40c848: _handle_and_close_when_done(<bound method ModbusServer.handle of <conpot.proto, <bound method StreamServer.do_close of <StreamServ, (<gevent._socket3.socket [closed]  object, fd=-1, )> failed with error
```

glaslos, 2020-10-19:
Did you get this error when you did a request to Conpot or did you see it in the wild? If you are doing the request, the client didn't send a payload. If this is in the wild, same reason. We could safe-guard against it.

M4rdc0re, 2024-07-17:
[image]

M4rdc0re, 2024-07-17:
Hi,
I'm having issues with modbus. The function struct.unpack is failing.
[image]
[image]

glaslos, 2026-09-14:
Guardrail added in b1cef75a9ff709e686cb43ebb27f41268e5570e7

closed 2026-09-14 by glaslos