mushorg/conpot #520: Conpot should not depend on external service

srenfo, 2020-11-02:
The SNMP protocol should not use a hardcoded server to fetch MIBs. In fact, downloads at runtime should probably be avoided altogether.

The offending lines are:

https://github.com/mushorg/conpot/blob/ff09e009d10d953aa7dcff2c06b7c890e6ffd4b7/conpot/protocols/snmp/command_responder.py#L62-L64

This has caused issues in the past (#516, possibly #510) when snmplabs.com went down (https://github.com/etingof/pysnmp/issues/376) and will cause issues again if/when Cisco change anything with their FTP server.

Ideally we would bundle the MIB sources with the templates. There are various repositories that they can be sourced from, including [net-snmp on GitHub](https://github.com/net-snmp/net-snmp/tree/master/mibs), or even the Cisco server from above. But I do not know what this would involve license-wise or copyright-wise, so I hesitate to create such a PR.

The other option is to install an appropriate package via `apt` (or `pip`?). There exists such a package for Fedora in [net-snmp-libs](https://fedora.pkgs.org/33/fedora-x86_64/net-snmp-libs-5.9-2.fc33.x86_64.rpm.html) (the MIBs are put in `/usr/share/snmp/mibs/` and get picked up by PySNMP), but I have failed to find one for Debian.

The cop-out answer is to move the server path to the template or config file, but since we would presumably set a default, this would still lead to complaints once the default stops working.

glaslos, 2020-11-13:
We could also drop `get_ext_ip`, I don't recall exactly why needed this in the first place :thinking:

srenfo, 2020-11-13:
> We could also drop `get_ext_ip`, I don't recall exactly why needed this in the first place thinking

git blame led me to #40 (there's a commit linked in the issue). :sunglasses: 

I'm all for removal if it's not needed. In the code it's only used in the `LogWorker`. It's being added to every emitted event. Maybe someone who currently runs Conpot in production (@t3chn0m4g3?) can chime in as to its usefulness?

We can always do a [Scream Test](http://v-wiki.net/the-scream-test/), i. e. remove it and see if anyone screams. (I just now learned that that has a name. :laughing:)

glaslos, 2020-11-13:
Yeah, this was from back in the day when we assumed people would setup sensors and contribute to the hpfeeds network :)

glaslos, 2020-11-15:
I don't think @t3chn0m4g3 is using MySQL, he usually collects data using hpfeeds or structured logs IIRC

t3chn0m4g3, 2020-11-15:
@glaslos Correct, not using MySQL. WRT `get_ext_ip` this is probably interesting for folks who run Conpot in a docker container to log the external `dest_ip`.

glaslos, 2026-09-14:
Closing this as the FTP dependency is gone and the default template is using pre-compiled MIBs. Since we allow templates to have custom MIBs, it would be impossible to anticipate shipping all MIB depencencies with Conpot.

closed 2026-09-14 by glaslos