mushorg/conpot #521: Dockerfile should build current code

srenfo, 2020-11-02:
Despite a very misleading comment, the Dockerfile will always build the public `master` rather than the code in the current working copy:

https://github.com/mushorg/conpot/blob/ff09e009d10d953aa7dcff2c06b7c890e6ffd4b7/docker/Dockerfile#L6-L9

If Docker is the recommended way to run Conpot then it should be possible to do so during development.

srenfo, 2020-11-02:
After an (admittedly quick) attempt at fixing this I'm not sure anyone actually uses the Dockerfile in production. It appears broken for the current code even beyond #519.

If the Docker approach is abandoned I suggest it is deleted rather than fixing this issue. If it is not I highly suggest it is incorporated into the CI pipeline (cf. #486).

Though again, I only had a quick go at this, so the error may well be on my end.

glaslos, 2020-11-03:
No, I can confirm the Dockerfile is borked. I'm working on a new one. I'll create a draft PR tomorrow. I'm a bit stuck or just need a 2nd pair of eyes...

glaslos, 2020-11-03:
See here: https://github.com/mushorg/conpot/pull/522

closed 2020-11-13 by glaslos