mushorg/conpot #523: Replace pycrypto with cryptography

glaslos, 2020-11-04:
Pycrypto is deprecated and also lacks a wheel. Replace with https://cryptography.io/en/latest/hazmat/primitives/symmetric-encryption.html

AlvinKuruvilla, 2020-11-11:
I want to take a stab at this do you think you could list the files that use pycrypto? Just so it be easier for me and any other people who want to assist

AlvinKuruvilla, 2020-11-11:
Thanks, I will take a look

AlvinKuruvilla, 2020-11-11:
@glaslos I'm having an issue with the `decrypt()` function in pycrypto and I wanted to see if you had any suggestions

So in the code currently this is how a new AES cipher is made:
`decrypter = AES.new(self.aeskey, AES.MODE_CBC, iv)`

I believe I have matched that functionality with:
`backend = default_backend()`
`cipher = Cipher(algorithms.AES(self.aeskey), modes.CBC(iv), backend=backend)`
Which returns a `Cipher` Object (which a least for the time being I am assuming is analogous to `AESCipher`

My problem lies with this code snippet where it is calling `decrypt()` on the `AESCipher` object
`
decrypted = decrypter.decrypt(
                    struct.pack("%dB" % len(payload[16:]), *payload[16:])
                )
`
I have been searching around and I'm unable to find an implementation of this decrypt function in the Github repo. The decrypt function as called here returns `bytes`

When using the cryptography module's decryptor function, however, it returns a `CipherContext` object but the subsequent lines expect bytes. Do you think it is as simple as casting the entire function call to `bytes` and adding in the same pass parameter that the pycrypto `decrypt()` function is taking in now?

Something like this?

`
decryptor = bytes(cipher.decryptor(
                    struct.pack("%dB" % len(payload[16:]), *payload[16:])
                ))
`

Sorry for the long-winded explanation, I was just trying to be thorough

glaslos, 2020-11-13:
@AlvinKuruvilla did you try the example? https://cryptography.io/en/latest/hazmat/primitives/symmetric-encryption.html#cryptography.hazmat.primitives.ciphers.Cipher

NishantPuri99, 2021-01-10:
I have a fair knowledge of Python, may I try this one out?

glaslos, 2021-01-10:
Sure, go aheard.

NishantPuri99, 2021-01-11:
Thanks @glaslos I'll get working with this in a few days

CypherpunkSamurai, 2021-12-08:
Hello, I wanted to help contribute to this project and I happen to look upon this issue.

> Pycrypto is deprecated and also lacks a wheel. Replace with https://cryptography.io/en/latest/hazmat/primitives/symmetric-encryption.html
I will try replacing pycrypto with cryptography, also can we use (should I try using) pycryptodome instead as a second option?

glaslos, 2022-02-20:
@CypherpunkSamurai have you had a chance to give this a try?

CelticLite, 2023-03-28:
Hey @glaslos, it looks like this issue has not been handled. Any chance I could open a PR with the fix? I have the code changes complete locally.

glaslos, 2023-03-29:
Absolutely 😊

CelticLite, 2023-03-29:
It looks like I do not have permissions to push a new branch to the repo:
`remote: Permission to mushorg/conpot.git denied to CelticLite.`
I followed the guide here: https://conpot.readthedocs.io/en/latest/development/guidelines.html

glaslos, 2023-03-29:
Yes, you need to make a fork and then create the PR from your for to this repo. No worries, GitHub will guide you :)

CelticLite, 2023-03-29:
PR is up and ready for review:  https://github.com/mushorg/conpot/pull/582

glaslos, 2023-07-24:
Closed with https://github.com/mushorg/conpot/pull/582
Thank you for your contribution!

closed 2023-07-24 by glaslos