mushorg/conpot #524: s7 read and write

Han-zb, 2020-11-04:
conpot didn't implement read and write and other methods in the s7 protocol, but i want to  communicate with conpot using s7 protocal. 
can anyone give me some tips on how to implement these methods? for example how to realize the db blocks, and what other code i need to add into the conpot?  i'm wondering if the snap7 library would help. 
thanks a lot!

soso288, 2021-04-13:
Hi were you able to implement something? If so would you mind sharing your results with me? :)

glaslos, 2026-09-14:
It is possible to implement this with inspiration from https://github.com/gijzelaerr/python-snap7, specifically https://python-snap7.readthedocs.io/en/latest/API/server.html 
Ideally this would also make use of the existing Databus in the core module.

closed 2026-09-16 by glaslos