mushorg/conpot #534: Replacing XML for templates

glaslos, 2020-11-15:
For #525 I had to write some XML which was rather unpleasant... I'm suggesting to replace our use of XML for defining the templates with something more human readable, suggestions welcome. 
I would definitely like to maintain the ability to verify a template with a schema. Whatever we decide to replace XML with, doesn't need to support a schema. but we can [verify the JSON](https://pypi.org/project/schema/) we parse from the template.

srenfo, 2020-11-16:
I would steer clear of JSON [1][2] and YAML [3], except maybe [StrictYAML](https://github.com/crdoconnor/strictyaml) [4]. Googling for "JSON configuration" had two articles advising against it in the first four results for me. You can find plenty about YAML issues and gotchas as well. I have had minor (JSON) to medium (YAML) issues using them as configuration myself.

On the other hand, with [PEP 518](https://www.python.org/dev/peps/pep-0518/) (`pyproject.toml`), [TOML](https://github.com/toml-lang/toml) has become a first-class citizen in Python. Contrary to JSON and YAML, it was explicitly designed as a configuration language. The PEP [discusses](https://www.python.org/dev/peps/pep-0518/#file-format) TOML and even includes a [comparison table](https://www.python.org/dev/peps/pep-0518/#other-file-formats) of TOML, JSON, YAML and INI. JSON Schema Everywhere has a [section about TOML](https://json-schema-everywhere.github.io/toml).

The articles above/below list [Hjson](https://hjson.github.io/), [JSON5](https://json5.org/) and [HOCON](https://github.com/lightbend/config/blob/master/HOCON.md) as further alternatives.

I do not have much experience with either TOML or the other alternatives, but just from the looks of it I suggest TOML.

[1] https://www.arp242.net/json-config.html
[2] https://www.lucidchart.com/techblog/2018/07/16/why-json-isnt-a-good-configuration-language/
[3] https://www.arp242.net/yaml-config.html
[4] https://news.ycombinator.com/item?id=17358103 (thread with the creators of YAML and StrictYAML weighing in, with some noise)

glaslos, 2020-11-16:
Yeah, I share your experience. There are options which are more structure (e.g. HCL) but I'm also leaning towards something which is easy to understand and familiar. I keep this issue around for a bit longer before making a decision. Maybe @t3chn0m4g3 or @xandfury have some input as well.

glaslos, 2026-09-14:
Update: [#637](https://github.com/mushorg/conpot/pull/637) introduces TOML dual-format loading with TFTP as the reference migration (matching the TOML lean in this thread).

Remaining protocol/profile cutover is tracked in a follow-up issue so this ticket stays the design home until XML can be deprecated.

glaslos, 2026-09-20:
TOML + schema validation is now the only template format for shipped profiles (follow-up #638 complete). Closing.

closed 2026-09-20 by glaslos