mushorg/conpot #535: config with hpfeeds

zainalawaludin, 2020-11-16:
where location config file if i want to send data to hpfeeds? FYI i install conpot with docker pull honeynet/conpot. Thanks

glaslos, 2020-11-16:
When you run the container Conpot will tell you which configuration was chosen (close to the top of the output). You can then override this config with `docker run -v your_config.cfg:/LOCATION_OF_THE_CFG...` replace `your_config.cfg` with the modified config you want to use and `LOCATION_OF_THE_CFG...` what you see in the containers output from where it get's the config. Could be:

`docker run -it -v config.cfg:/home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/testing.cfg honeynet/conpot`

Let me know if that works for you.

zainalawaludin, 2020-11-17:
hi @glaslos can u share code of config how send data from conpot to hpfeeds? because i have create config file with activated hpfriends but doesn't work :

[hpfriends]
enabled = True
host = 117.53.44.90
port = 10000
ident = isifconpot
secret = password1234
channels = ["conpot.events", ]

closed 2020-11-18 by zainalawaludin