mushorg/conpot #536: Flatten template directory structure?

srenfo, 2020-11-16:
For simplifications beyond #534 the template directories could be flattened a bit.

Currently:

```
conpot/templates
├── default
│   ├── bacnet
│   │   └── bacnet.xml
│   ├── enip
│   │   └── enip.xml
│   ├── ftp
│   │   └── ftp.xml
│   ├── http
│   │   ├── htdocs
│   │   │   ├── tests
│   │   │   └── ...
│   │   ├── statuscodes
│   │   │   └── ...
│   │   └── http.xml
│   ├── ipmi
│   │   └── ipmi.xml
│   ├── ...
│   └── template.xml
...
```

Suggestion: Move the protocol XMLs up by one level. Keep the subdirectories for protocols that want auxiliary files.

```
conpot/templates
├── default
│   ├── bacnet.xml
│   ├── enip.xml
│   ├── ftp.xml
│   ├── http
│   │   ├── htdocs
│   │   │   ├── tests
│   │   │   └── ...
│   │   ├── statuscodes
│   │   │   └── ...
│   ├── http.xml
│   ├── ipmi.xml
│   ├── ...
│   └── template.xml
...
```

Pros: Easier to navigate and thus easier to modify and to keep track of.

Cons: It may look odd to have both an `http.xml` and an `http` directory. Also, `template.xml` is now on the same level as the protocol XMLs despite being one level above them hierarchically.

glaslos, 2020-11-16:
How about stuffing all protocols into a flat `protocol` folder with `http` as suggested? Kind of a middle ground?

srenfo, 2020-11-20:
Do you mean their XMLs or their auxiliary data?

Either way this is not a big deal, but we were talking about simplifying things anyway :see_no_evil:

glaslos, 2020-11-21:
This was my suggestion. It would keep the visibility for the template.xml
```
conpot/templates
├── default
│   ├── protocols
│   │   ├── bacnet.xml
│   │   ├── enip.xml
│   │   ├── ftp.xml
│   │   ├── http
│   │   │   ├── htdocs
│   │   │   │   ├── tests
│   │   │   │   └── ...
│   │   │   ├── statuscodes
│   │   │   │   └── ...
│   │   ├── http.xml
│   │   ├── ipmi.xml
│   │   ├── ...
│   └── template.xml
```

closed 2026-09-13 by glaslos