mushorg/conpot #537: Description of default template no longer accurate

srenfo, 2020-11-16:
The current help text of the `default` template is outdated and wrong:

```
--------------------------------------------------
 Available templates:
--------------------------------------------------
[...]
   --template default
       Unit:        Siemens - S7-200
       Desc:        Rough simulation of a basic Siemens S7-200 CPU with 2 slaves
       Protocols:   HTTP, MODBUS, s7comm, SNMP
       Created by:  the conpot team
[...]
```

For reference:

```sh
$ tree -d
.
├── bacnet
├── enip
├── ftp
├── http
│   ├── htdocs
│   │   └── tests
│   └── statuscodes
├── ipmi
├── modbus
├── s7comm
├── snmp
├── ssl
└── tftp
```

(`ssl` is not a protocol)

This does not strike me as an S7-200 any more. IMO we could/should

- [ ] Rewrite the description
- [ ] Check the other templates for outdated descriptions
- [ ] Derive the "Protocols" section from the filesystem rather than try to keep filesystem and XML in sync
- [ ] Restore the original intent under a different template name by recreating an S7-200-like template from the git history
- [ ] Rename the `default` template, maybe to `everything`? Though that is not quite true since the guardian_ast and kamstrup (and Proxy, technically) protocols aren't enabled. Maybe change that as well?

I don't know how important say the S7-200 template is. Maybe open separate issues for each?

glaslos, 2020-11-16:
Haha, I was about to create a new ticket for the template description :joy: 
how about renaming `default` to `sample`? IIRC that was the intention, to show a basic configuration for all the protocols available.

srenfo, 2020-11-20:
Sure. Changing the `Desc:` text to match should be non-controversial then if that was the original intention anyway. :smiley_cat: 

Is there any value in restoring an S7-200 template? I don't know how close to reality that template was to begin with or how much demand there might be.

southwestflavorz, 2022-09-30:
Is there a repo of common templates folks are using? I'm a student who's testing honeypot interactions for analysis, and I would love to include additional ICS devices beyond the S7-200 that serves as the "default"

closed 2026-09-13 by glaslos