mushorg/conpot #538: The docker container should use a specific config

glaslos, 2020-11-16:
Currently we run the container with `-f` forcing to use the testing config. This is rather problematic...
Create a new `docker.cfg` with sane defaults and modify the container to use this configuration.
Add documentation on how one can use `docker run -v config.cfg:config.cfg` to modify the containers configuration of Conpot.

NishantPuri99, 2021-01-10:
Hi, Can I try this one? Could you please provide me a direction for it?

glaslos, 2021-01-10:
Can you explain what you have done so far and where you are stuck? Is there something unclear from the ticket?

NishantPuri99, 2021-01-11:
I just started out with this issue. I'm currently trying #523 and #540 . I'll try this out as soon as I've done the other two. Thank you

HiDomesticCat, 2024-08-05:
Hello,

I'd like to ask if anyone is currently working on this issue. If not, I'd like to give it a try. However, I'm still a beginner and would like to know where I can start.

glaslos, 2024-08-05:
Not to my knowledge. Looking forward to see what you come up with. Let me know if you need help

HiDomesticCat, 2024-08-07:
I would like to clarify my understanding of the issue and would appreciate your confirmation on whether it meets the requirements.

I modified the `CMD` directive in the Dockerfile from `-f` to `-c /config.cfg`. I have two questions regarding this change:

1. Is it necessary to allow users to switch back to `-f`?
2. I placed `config.cfg` in the root directory. Is this considered good practice? I thought it might make the `docker run` command cleaner.

Additionally, I created a new `docker.cfg` file in the conpot directory and a `dockerTest` directory for filesystem mounting. Is this approach correct?

After these modifications, users need to run `docker run -v ./conpot/docker.cfg:/config.cfg ...` (it will result in an error if -v is not included).

I apologize if my questions seem trivial. Please let me know if you need any further information. Thank you.

closed 2026-09-06 by glaslos