mushorg/conpot #539: Non of the quick installation methods work

lukacsnikolett, 2020-11-21:
The documentation mentions 3 ways to install conpot but non of them works. Building it is not possible either way, the 7. step of the Dockerfile fails. The pre-build image starts, but when I run conpot it says 'conpot not found'.
How can I start this thing? Please update the docs with a method that is actually useful and isn't wasting my time.

glaslos, 2020-11-21:
This tone is not appreciated. We work on Conpot in our free time and make it available for free.

glaslos, 2020-11-21:
I just build the container from the latest Dockerfile. It worked for me. My only guess is that you or a dependency had a network issue during the installation of the dependencies (step 7). If you would provide more details on how it fails, I could give you more help.

glaslos, 2020-11-21:
The pre-build conpot can be run with `/home/conpot/.local/bin/conpot -f --template default`

closed 2020-12-06 by glaslos