mushorg/conpot #540: Refactor bin/conpot

glaslos, 2020-11-22:
* Would be nice if it can be a file conpot.py with Python extension. setup.py needs to be updated to make this possible.
* The file should only do flags, startup message and initializing/running Conpot.
* Everything template should be in a separate module.
* Everything starting protocols should be in a separate module.
* Everything logging should be in separate module.

NishantPuri99, 2021-01-10:
Hi, Can I try this out?

glaslos, 2021-01-10:
Of course, no need to ask :)

srenfo, 2021-01-30:
Ideally there would also be tests where possible. Don't know if you want to add this to the requirements.

I realize that testing `main()` is notoriously difficult, but with a split in separate modules (or at least functions) this should be much easier.

glaslos, 2022-02-20:
@NishantPuri99 have you had any success?

glaslos, 2026-09-05:
Closed with https://github.com/mushorg/conpot/pull/632

closed 2026-09-05 by glaslos