mushorg/conpot #552: Problem scanning Conpot default template for S7comm protocol

cyberitot79, 2021-02-22:
Hello,

I wanted to ask for a problem I'm having using Conpot. I've been trying to use the honeypot to simulate PLCs devices using Modbus and S7 protocols. But I'm having some problems with S7 protocol, even with the default templates, not with the ones I modified, so I'm opening this issue for the case you could help me.

I've read other previous issues, and I've read so much time the code trying to understand it. If I'm not wrong, Conpot should give support against tools or scripts like plcscan (https://github.com/meeas/plcscan). But even using the default template without any modification, I find no results for this plcscan scanning. As they say in their github info, the results should be similar to something like this when scanning s7:

**127.0.0.1:102 S7comm (src_tsap=0x100, dst_tsap=0x102)
   Module                   : 6ES7 151-8AB01-0AB0  v.0.2    	(36455337203135312d38414230312d304142302000c000020001)
   Basic Hardware           : 6ES7 151-8AB01-0AB0  v.0.2    	(36455337203135312d38414230312d304142302000c000020001)
   Basic Firmware           :                      v.3.2.6  	(202020202020202020202020202020202020202000c056030206)
   Unknown (129)            : Boot Loader           A       	(426f6f74204c6f61646572202020202020202020000041200909)
   Name of the PLC          : SIMATIC 300(xxxxxxxxx)        	(53494d4154494320333030280000000000000000002900000000000000000000)
   Name of the module       : IM151-8 PN/DP CPU             	(494d3135312d3820504e2f445020435055000000000000000000000000000000)
   Plant identification     :                               	(0000000000000000000000000000000000000000000000000000000000000000)
   Copyright                : Original Siemens Equipment    	(4f726967696e616c205369656d656e732045717569706d656e74000000000000)
   Serial number of module  : S C-BOUVxxxxxxxx              	(5320432d424f5556xxxxxxxxxx00000000000000000000000000000000000000)
   Module type name         : IM151-8 PN/DP CPU             	(494d3135312d3820504e2f445020435055000000000000000000000000000000)**

But when I use the script to test that, I get no answers (as you can see in the image I'm attaching to the issue, that's my output result). I know using default conpot would never give me the same answer as they say in plcscan github, as Conpot doesn't have all values for all the fields (basic hardware, firmware...). But at least I guess it should give me the ones it has, as I've seen in other people using conpot and plcscan. But I'm not sure if this is because of Conpot (TODAY) gives no support to this kind of tool (plcscan) or if there's something wrong, as I'm using the default templates and code, I did not change it to test this.

If it helps, I'm using Ubuntu 18.04 LTS. I've tried this in other machines, or using docker, and always the output is the same, with no results. I have all libraries installed as the script works with other conpot instances from other people. I'm testing it against localhost and 102, as it's where it's listening.

Thank you for your time and for your help.
[image]

glaslos, 2021-03-01:
Hi @cyberitot79 , do you see any output in Conpot when you do the scan?

cyberitot79, 2021-03-02:
Hi!

I know it's at least connecting to the port as it seems to create Conpot output. Next picture shows what it logs when I do the scan:
[image]

glaslos, 2021-04-16:
Please provide text instead of images to make it searchable and more accessible.
I'm not too familiar with SNMP, did you check if we support the PDU type 7?

soso288, 2021-04-23:
I have the same problem. After looking at the source code I couldn't find a solution as the information is received by PLCscan (found out during debugging), but not displayed. I think it has got to do with incorrect packing or unpacking of the data. Where you able to solve the problem?