mushorg/conpot #553: I modified Conpot's default template.xml file to mimic a PowerLogic meter. However, I get this error while trying to run it. Any help is most welcome and will acknowledge the same in my thesis.

WZcabral, 2021-03-01:
[image]

glaslos, 2021-03-01:
@WZcabral it would be great if you could paste the text from the error instead of providing an image. This makes it possible for others and search engines to find your described issue in the future.
Your configuration is not matching the [schema](https://github.com/mushorg/conpot/blob/master/conpot/tests/template_schemas/core.xsd). You should be able to validate it using the linked schema. 
If you provide your config, I might be able to help you spot the issue.

glaslos, 2021-03-01:
@WZcabral provided the template in private. It seems you can solve your issues by removing the outer `conpot_template`. Your template should look like:
```
<core>
    <template>
...
```

WZcabral, 2021-03-01:
Thanks, @glaslos! I did just that however, now I am getting a new error which is: 

```
WARNING:root:--force option specified. Using testing configuration
2021-03-01 02:44:11,866 --force option specified. Using testing configuration
2021-03-01 02:44:11,866 Starting Conpot using template: /usr/local/lib/python3.6/dist-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default
2021-03-01 02:44:11,867 Starting Conpot using configuration found in: /usr/local/lib/python3.6/dist-packages/conpot-0.6.0-py3.6.egg/conpot/testing.cfg
ERROR:root:Error parsing XML template: /usr/local/lib/python3.6/dist-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default/template.xml:66:0:ERROR:SCHEMASV:SCHEMAV_ELEMENT_CONTENT: Element 'protocols': This element is not expected.
2021-03-01 02:44:11,867 Error parsing XML template: /usr/local/lib/python3.6/dist-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default/template.xml:66:0:ERROR:SCHEMASV:SCHEMAV_ELEMENT_CONTENT: Element 'protocols': This element is not expected.
```
So, I deleted the entire protocol section in the template and edited the protocols separately within their own files which seemed to fix the issues.

closed 2021-03-01 by WZcabral