mushorg/conpot #554: Json logging is creating empty request and response

mintos5, 2021-03-05:
Hi, I am using your honeypot from T-pot platform: https://github.com/telekom-security/tpotce and it is working really well.
I just found a small bug connected with logging to JSON files.
Fields **request** and **response** have a lot of time a value of type **bytes** which is not properly decoded to JSON. This can be easily fixed by adding instructions on how to decode bytes to string in function json_default(obj):
https://github.com/mushorg/conpot/blob/804fd65aa3b7ffa31c07fd4e863d4a5500414cf3/conpot/core/loggers/helpers.py#L5-L11

I hope you can reproduce this bug and find a fix for it.

glaslos, 2021-04-16:
Hey, would you mind providing a PR with your fix?

glaslos, 2022-02-20:
@mintos5 I've merged your fix. Could you confirm your issue is resolved?

mintos5, 2022-03-03:
Thanks for merging the fix. I can confirm that the issue is resolved.

closed 2022-03-03 by mintos5