mushorg/conpot #556: service conpot down after  being attack like ddos

zainalawaludin, 2021-03-16:
hi i have problem with my conpot, if there are attack like  Ddos attack, my conpot cannot access from browser, and finally i must restart docker conpot to resolve this problem. below capture from attacker 
[image]
any idea for this problem? thanks

glaslos, 2021-04-16:
Hi, please provide the log as text instead of an image. It makes it searchable and more accessible.
I don't think we can do much more in this case. The SNMP server is rejecting the requests but even that requires resources which seem to get exhausted. We could potentially be faster at discarding the requests but I don't think we will win this game. Do you have any chance to prevent this before it reaches Conpot?

glaslos, 2022-02-20:
Closing due to inactivity.

closed 2022-02-20 by glaslos