mushorg/conpot #557: Implementing a custom web interface

soso288, 2021-04-16:
I want conpot to be more realistic and to provide more interaction to a potential attacker. Therefore I tried implementing my own webpage and put it under .../site-packages/conpot/templates/s7-300/http/htdocs/index.html. 
When I open the page in my browser it can't be loaded and I get this error in conpot:
[image]
Can somebody help me understand it? I use conpot with virtualenv on Ubuntu 18.04

glaslos, 2021-04-16:
Hi, please provide the error message as text, it makes it searchable and more accessible.
It seems to be an encoding issue in the html file. Either try to find the issue (character in position 594) or modify the Conpot to ignore encoding errors.

soso288, 2021-04-18:
Thank you I was able to solve the problem. I have another two questions: 

1. How can I modify the s7comm.xml file so that I can view the module and component information with plcscan? Plcscan always completes without displaying anything.
2. How can I modify the modbus.xml so that plcscan gives other output than slave device failure when scanning the honeypot?

Thanks for your help!

closed 2021-05-06 by soso288