mushorg/conpot #561: Getting error while testing modbus

prashantvidja, 2021-08-03:
**Describe the bug**
When I try to test Modbus getting the below error.

> ERROR:conpot.protocols.modbus.slave:Exception caught: Modbus Error: Exception code = 2. (A proper response will be sent to the peer)
> 2021-08-03 05:56:54,930 Exception caught: Modbus Error: Exception code = 2. (A proper response will be sent to the peer)
> Traceback (most recent call last):
>   File "src/gevent/greenlet.py", line 906, in gevent._gevent_cgreenlet.Greenlet.run
>   File "/home/pr/conpot/lib/python3.6/site-packages/gevent/baseserver.py", line 34, in _handle_and_close_when_done
>     return handle(*args_tuple)
>   File "/home/pr/conpot/lib/python3.6/site-packages/conpot/protocols/modbus/modbus_server.py", line 125, in handle
>     query, request, self.mode
>   File "/home/pr/conpot/lib/python3.6/site-packages/conpot/protocols/modbus/slave_db.py", line 80, in handle_request
>     return (None, {'request': request_pdu.encode('hex'),
> AttributeError: 'bytes' object has no attribute 'encode'
> 2021-08-03T05:56:54Z <Greenlet at 0x7f1fb4112448: _handle_and_close_when_done(<bound method ModbusServer.handle of <conpot.proto, <bound method StreamServer.do_close of <StreamServ, (<gevent._socket3.socket [closed] at 0x7f1fb40a4db)> failed with AttributeError

**To Reproduce**
Steps to reproduce the behavior:
1. Test the Modbus connection with the below script.
```python
from pymodbus.client.sync import ModbusTcpClient
client = ModbusTcpClient('127.0.0.1',5020)
client.write_coil(1, True)
result = client.read_coils(2,2)
print(result.bits[0])
client.close()
```
2. you can find errors in the console.

**Expected behavior**
It should work without any exception.

**Desktop (please complete the following information):**
 - OS: Linux
 - Version Ubuntu 18.04

**Additional context**
I have tested on the latest 0.6.0 version.

glaslos, 2022-02-20:
Hey, I have pushed a fix for the error you encountered.
In order to use pymodmus, you have to provide a slave ID: `client.write_coil(1, True, unit=0x01)`

prashantvidja, 2022-02-21:
Hey @glaslos 

Thanks for the update.
Will test it out.