mushorg/conpot #564: Conpot stops functioning after nmap scans

bestrocker221, 2022-01-03:
Conpot does not work anymore after nmap scanning. Both through the container and not.
The process keeps running, but the ports do not react anymore and the logging is stopped.

This unfortunately makes the honeypot pretty much useless since it cannot withstand a basic nmap scan.

**To Reproduce**

1. `docker-compose up`
2. `nmap -sV -v 127.0.0.1`
3. see few logs and maybe the scan terminates.
4. scan can also not terminate, and conpot is unresponsive, all future port probes will return no answer. No further logs.

**Expected behavior**
Conpot keeps running, with the logs saved of each connection attempt.

Any help?

Thank you!

Vingaard, 2022-01-04:
Interesting - a quick question. what happens if you scan the Conpot from anything else than 127.0.0..1 (localhost) 
like nmap scan: 
nmap -sV -v {inset conpot_ip}

bestrocker221, 2022-01-04:
Same exact behavior.

bestrocker221, 2022-01-25:
Anybody confirming the same behaviour or found an alternative way to solve it?

bestrocker221, 2022-02-19:
Hi again.
How are you supposed to use conpot if it can't withstand the most basic nmap scan without crashing and freezing?
Am I missing something?

Tried again on a fresh Debian 11 VM, with normal installation and default template.

1. `conpot -c /home/user/prod1.cgf --template default`
2. from the host: `nmap CONPOT-VM-IP -sV`
3. conpot logging freezes and the process must be restarted. When ctrl+c the process, many errors on `sock.recv` pops out (event._gevent_cgreenlet.Greenlet), like if the logging daemon is stuck waiting for all the connections to terminate correctly? or some buffer gets overflowed? 

Any help? I find it hard to believe that conpot is expected to behave like this, the same moment it is deployed on the Internet it is going to crash and need restarting every 5 minutes.

Thank you

glaslos, 2022-02-20:
I've just checked out the latest code from GitHub, run `docker-compose` up and then `nmap -sV -v 127.0.0.1` twice without issues. CPU usage was pretty much at 100% during the nmap scan. Scan took about 150 seconds.

bestrocker221, 2022-02-21:
Strange, I tried both the docker-compose and through the virtualenv, on fresh new Debian and Arch systems and I got the explained issue.

I found many places in the code where the issue makes sense, the lack of sockets closing and error handling during loops. Gonna make a PR when finished. Now it seems to deterministically work

t3chn0m4g3, 2022-03-09:
@glaslos 

Built `conpot` from latest master and can now confirm that problems with `iec104` are **fixed**, while `guardian_ast` repeatedly shows 100% CPU usage after `nmap`.
[image]

It is reproducible with `nmap -p 10001 -sV -v 127.0.0.1`.

Console log:
```
2022-03-09 14:02:37,741 Config file found!
2022-03-09 14:02:37,743 Starting Conpot using template: /usr/lib/python3.9/site-packages/conpot/templates/guardian_ast
2022-03-09 14:02:37,743 Starting Conpot using configuration found in: /etc/conpot/conpot.cfg
2022-03-09 14:02:37,750 Serving tar:///usr/lib/python3.9/site-packages/conpot/data.tar as file system. File uploads will be kept at : /tmp/conpot
2022-03-09 14:02:37,750 Opening path /tmp/conpot for persistent storage of files.
2022-03-09 14:02:37,751 Initializing Virtual File System at /tmp/conpot/__conpot__hi0s02wj. Source specified : tar:///usr/lib/python3.9/site-packages/conpot/data.tar
 Please wait while the system copies all specified files
2022-03-09 14:02:37,818 Fetched 1.2.3.4 as external ip.
2022-03-09 14:02:37,819 Conpot GuardianAST initialized
2022-03-09 14:02:37,819 Found and enabled guardian_ast protocol.
2022-03-09 14:02:37,820 No proxy template found. Service will remain unconfigured/stopped.
2022-03-09 14:02:37,820 GuardianAST server started on: ('0.0.0.0', 10001)
2022-03-09 14:04:14,067 New guardian_ast session from 192.168.208.1 (7f0c9159-271d-432c-9e01-2c5eb1d75110)
2022-03-09 14:04:14,067 New GuardianAST connection from 192.168.208.1:60070. (7f0c9159-271d-432c-9e01-2c5eb1d75110)
2022-03-09 14:04:20,071 Non ^A command attempt 192.168.208.1:60070. (7f0c9159-271d-432c-9e01-2c5eb1d75110)
2022-03-09 14:04:20,071 GuardianAST client disconnected 192.168.208.1:60070. (7f0c9159-271d-432c-9e01-2c5eb1d75110)
2022-03-09 14:04:20,073 New GuardianAST connection from 192.168.208.1:60074. (7f0c9159-271d-432c-9e01-2c5eb1d75110)
```

JSON log:
```
{"timestamp": "2022-03-09T12:03:49.935986", "sensorid": "conpot", "id": "c1d7e4be-0a27-46fa-88f6-18d65f2deb1b", "src_ip": "192.168.128.1", "src_port": 52392, "dst_ip": "192.168.128.2", "dst_port": 10001, "public_ip": "1.2.3.4", "data_type": "guardian_ast", "request": null, "response": null, "event_type": "NEW_CONNECTION"}
{"timestamp": "2022-03-09T12:03:49.935986", "sensorid": "conpot", "id": "c1d7e4be-0a27-46fa-88f6-18d65f2deb1b", "src_ip": "192.168.128.1", "src_port": 52392, "dst_ip": "192.168.128.2", "dst_port": 10001, "public_ip": "1.2.3.4", "data_type": "guardian_ast", "request": null, "response": null, "event_type": "CONNECTION_LOST"}
{"timestamp": "2022-03-09T12:03:49.935986", "sensorid": "conpot", "id": "c1d7e4be-0a27-46fa-88f6-18d65f2deb1b", "src_ip": "192.168.128.1", "src_port": 52392, "dst_ip": "192.168.128.2", "dst_port": 10001, "public_ip": "1.2.3.4", "data_type": "guardian_ast", "request": null, "response": null, "event_type": "NEW_CONNECTION"}
{"timestamp": "2022-03-09T14:04:14.067252", "sensorid": "conpot", "id": "7f0c9159-271d-432c-9e01-2c5eb1d75110", "src_ip": "192.168.208.1", "src_port": 60070, "dst_ip": "192.168.208.2", "dst_port": 10001, "public_ip": "1.2.3.4", "data_type": "guardian_ast", "request": null, "response": null, "event_type": "NEW_CONNECTION"}
{"timestamp": "2022-03-09T14:04:14.067252", "sensorid": "conpot", "id": "7f0c9159-271d-432c-9e01-2c5eb1d75110", "src_ip": "192.168.208.1", "src_port": 60070, "dst_ip": "192.168.208.2", "dst_port": 10001, "public_ip": "1.2.3.4", "data_type": "guardian_ast", "request": null, "response": null, "event_type": "CONNECTION_LOST"}
{"timestamp": "2022-03-09T14:04:14.067252", "sensorid": "conpot", "id": "7f0c9159-271d-432c-9e01-2c5eb1d75110", "src_ip": "192.168.208.1", "src_port": 60070, "dst_ip": "192.168.208.2", "dst_port": 10001, "public_ip": "1.2.3.4", "data_type": "guardian_ast", "request": null, "response": null, "event_type": "NEW_CONNECTION"}
``` 

Happy to provide debug logs, just let me know what you need.

bestrocker221, 2022-03-09:
In my latest local version all the protocols in the default template works correctly after multiple `-sV` scans. I just need to find a proper way to make a PR since I am making modifications directly in my virtualenv dev environment.

t3chn0m4g3, 2022-03-09:
@bestrocker221 Interesting, `default` shows no sign of 100% CPU usage (after `nmap` is finished) on my end with latest master.

t3chn0m4g3, 2022-03-11:
Behavior is the same with Py 3.7/3.8/3.9.

Dirty, but works fine if `docker-compose` uses `restart=always`:
```
STOPSIGNAL SIGINT
HEALTHCHECK CMD if [ $(ps -C mpv -p 1 -o %cpu | tail -n 1 | cut -f 1 -d ".") -gt 90 ]; then kill -2 1; else exit 0; fi
```

- `-p 1` Monitors PID 1 only and kills process if process CPU usage is > 90%.

Keeping 🤞  for a solution, though 😄

bestrocker221, 2022-03-13:
@t3chn0m4g3 Hi, the solution might work but is just a workaround that does not actually fix the code behind it. 

I left conpot with default template running for a week and it eventually stopped functioning, now I checked and it was using 100% CPU, but since it is one single process I can't distinguish without a debugger what is going on.. need more testing.

Restarting the process as you say can work, but is indeed "dirty".

t3chn0m4g3, 2022-03-30:
@glaslos @xandfury Can you reproduce? Can I provide logs to help solving this?

bad-antics, 2026-02-01:
This is a known issue with ICS honeypots and aggressive nmap scanning. Here are some mitigations:

Stability Fixes:
1. Increase connection timeouts in conpot.cfg
2. Add rate limiting at firewall level: iptables -A INPUT -p tcp --syn -m limit --limit 10/s -j ACCEPT
3. Run conpot in a container with restart policy: docker run --restart=unless-stopped

For nmap detection:
- conpot should handle the probes gracefully, but aggressive scans (-T4, -T5) can overwhelm
- Consider using a reverse proxy like nginx in front for connection handling

Debug steps:
1. Check conpot logs during scan: tail -f /var/log/conpot.log
2. Monitor memory/CPU during scan
3. Try limiting concurrent connections in conpot config

Alternative: Run multiple conpot instances behind a load balancer for resilience.

What specific conpot configuration are you using? And which nmap flags trigger the issue?

glaslos, 2026-09-05:
Closed with https://github.com/mushorg/conpot/commit/1f50c7f1c0b782a0cb549ffb438373a1ab24cb51

closed 2026-09-05 by glaslos