mushorg/conpot #565: can not run conpot

K0zero, 2022-02-07:
ubuntu16.04 python3.6  
i use pip install the conpot instead of docker
but when i use conpot  the error came
Traceback (most recent call last):
  File "/usr/local/bin/conpot", line 4, in <module>
    __import__('pkg_resources').run_script('conpot==0.6.0', 'conpot')
  File "/home/zky/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 656, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/home/zky/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 1443, in run_script
    .format(**locals()),
pkg_resources.ResolutionError: Script 'scripts/conpot' not found in metadata at '/home/zky/.local/lib/python3.6/site-packages/conpot-0.6.0.dist-info'
i have watched others' issues but their question were different of mine 
Script 'scripts/conpot' not found in metadata at '/home/zky/.local/lib/python3.6/site-packages/conpot-0.6.0.dist-info'

glaslos, 2022-02-20:
Can you provide the exact commands you use?
I just tried successfully with:
```
pip install conpot
conpot
```

K0zero, 2022-02-22:
yes，i only used the command -- pip3 install conpot
by the way,  i did not use the root.

Harshiths1, 2023-10-19:
Try the following command after pulling from docker as i have also tried many times using pip install and pip3 still does not work but tried using ```
sudo docker run -it -p 81:81 -p 103:103 -p 503:503 -p 160:160/udp --network=bridge
honeynet/conpot
```
and it worked so let me know if it works for you too

closed 2026-03-28 by glaslos