mushorg/conpot #572: Bacnet not working properly

bestrocker221, 2022-04-23:
Default template, when scanning with nmap bacnet-info (for example) in debug mode, it throws an error saying its unable to read vendor ID.
Seems like bacnet connections just start, then end.

**To Reproduce**
Steps to reproduce the behavior:
1. Run conpot default template
2. `sudo nmap --script bacnet-info -sU -p 47808 127.0.0.1 -d`

**Expected behavior**
_Something_ like this
```
47808/udp open  bacnet
| bacnet-discover:
|   Vendor ID: BACnet Stack at SourceForge (260)
|   Vendor Name: BACnet Stack at SourceForge
|   Instance Number: 260001
|   Firmware: 0.8.2
|   Application Software: 1.0
|   Object Name: SimpleServer
|   Model Name: GNU
|   Description: server
|_  Location: USA
```
**Result**
```
New bacnet session from 127.0.0.1 (e88d8afd-0024-40ce-83b1-8d683fd121cc)
New Bacnet connection from 127.0.0.1:58919. (e88d8afd-0024-40ce-83b1-8d683fd121cc)
Bacnet PDU received from 127.0.0.1:58919. (ComplexAckPDU)
Bacnet client disconnected 127.0.0.1:58919. (e88d8afd-0024-40ce-83b1-8d683fd121cc)
New Bacnet connection from 127.0.0.1:58921. (e88d8afd-0024-40ce-83b1-8d683fd121cc)
Bacnet PDU received from 127.0.0.1:58921. (ComplexAckPDU)
Bacnet client disconnected 127.0.0.1:58921. (e88d8afd-0024-40ce-83b1-8d683fd121cc)
Session timed out: 089f4c44-a6ba-431e-bde3-bc470e380a85
Session timed out: e88d8afd-0024-40ce-83b1-8d683fd121cc
```

```
NSE: Starting bacnet-info against 127.0.0.1:47808.                                                                          
Discovered open port 47808/udp on 127.0.0.1                   
NSE: [bacnet-info 127.0.0.1:47808] Error receiving Vendor ID: Invalid BACNet packet                                         
NSE: bacnet-info against 127.0.0.1:47808 threw an error!                                                                    
/usr/bin/../share/nmap/scripts/bacnet-info.nse:1297: attempt to perform arithmetic on a nil value (local 'value')
stack traceback:                              
        /usr/bin/../share/nmap/scripts/bacnet-info.nse:1297: in function </usr/bin/../share/nmap/scripts/bacnet-info.nse:129
2>                                            
        (...tail calls...)           
        /usr/bin/../share/nmap/scripts/bacnet-info.nse:1533: in function </usr/bin/../share/nmap/scripts/bacnet-info.nse:147
7>                                                                                           
        (...tail calls...)                                                                   
                                                                                             
Completed NSE at 14:05, 0.00s elapsed 
..
..
PORT      STATE SERVICE REASON                                
47808/udp open  bacnet  script-set  
..
..
```

mintos5, 2023-01-19:
I am having similiar problems with bacnet.
I checked the source code and found some problems with decoding Bacnet Data
It was not adding/reading BVLC and NPDU layers.
I tried to fix that with my fork of conpot:
https://github.com/mushorg/conpot/compare/master...mintos5:conpot:bacnet_fix

Maybe this code can help you and maybe in the future it can be added with pull request.

M4rdc0re, 2024-07-17:
Hi,

Did you do a pull request?

glaslos, 2026-09-05:
Fixed by https://github.com/mushorg/conpot/pull/631

closed 2026-09-05 by glaslos