mushorg/conpot #573: FTP not working, commands are async and random responses are returned

bestrocker221, 2022-04-24:
Hi, when connecting via FTP towards the default conpot configuration, commands and response are randomly returned.

**To Reproduce**
1. conpot default template
2. ftp <IP> 2121
3. `ls` multiple times

**Expected behavior**
```
ftp> ls
150 Here comes the directory listing.
-rwxrwxrwx   1 nobody   ftp            49 Aug 01  2018 ftp_data.txt
```

Received:
```
ftp> ls
226 Transfer complete.
200 PORT Command Successful. Consider using PASV.
ftp> ls
150 Here comes the directory listing.
ftp> ls
226 Directory send OK.
200 PORT Command Successful. Consider using PASV.
ftp> ls
150 Here comes the directory listing.
ftp> ls
226 Directory send OK.
226 Transfer complete.
ftp> ls
150 Here comes the directory listing.
ftp> ls
226 Directory send OK.
501 Invalid PORT format.
ftp> ls
200 PORT Command Successful. Consider using PASV.
150 Here comes the directory listing.
-rwxrwxrwx   1 nobody   ftp            49 Aug 01  2018 ftp_data.txt
226 Directory send OK.
ftp> ls
226 Transfer complete.
200 PORT Command Successful. Consider using PASV.
ftp> ls
150 Here comes the directory listing.
ftp> ls
226 Directory send OK.
501 Invalid PORT format.

```

**Desktop (please complete the following information):**
 - OS: Debian 11
 - kernel 5.16

srdjus, 2025-02-13:
I have the same problem.

Any solutions?

```
Password: 
220- Technodrome - Mouser Factory. Authorized personnel only
220 
Remote system type is UNIX.
Using binary mode to transfer files.
ftp> ls
227 Entering Passive Mode (127,0,0,1,151,119).
150 Here comes the directory listing.
-rwxrwxrwx   1 nobody   ftp            29 Aug 01  2018 secret.txt
-rwxrwxrwx   1 nobody   ftp            36 Aug 01  2018 users.txt
226 Directory send OK.
ftp> ls
226 Transfer complete.
wrong server: return code must be 227
ftp> ls
227 Entering Passive Mode (127,0,0,1,203,115).
227 Entering Passive Mode (127,0,0,1,143,203).
ftp> ls
550 PASV command failed.
500 Command 'LIST\r\nPASV' not understood
ftp: Can't bind for data connection: Address already in use
ftp> ls
200 PORT Command Successful. Consider using PASV.
150 Here comes the directory listing.
-rwxrwxrwx   1 nobody   ftp            29 Aug 01  2018 secret.txt
-rwxrwxrwx   1 nobody   ftp            36 Aug 01  2018 users.txt
226 Directory send OK.
ftp> ls
226 Transfer complete.
200 PORT Command Successful. Consider using PASV.
ftp> ls
150 Here comes the directory listing.
ftp> ls
226 Directory send OK.
501 Invalid PORT format.
ftp> ls
200 PORT Command Successful. Consider using PASV.
150 Here comes the directory listing.
-rwxrwxrwx   1 nobody   ftp            29 Aug 01  2018 secret.txt
-rwxrwxrwx   1 nobody   ftp            36 Aug 01  2018 users.txt
226 Directory send OK.
ftp>
```

glaslos, 2026-09-05:
Fixed with https://github.com/mushorg/conpot/commit/626a2e94b0652ee1e3d5cfb9253bde945886df3e

closed 2026-09-05 by glaslos