mushorg/conpot #576: conpot -f --template default /bin/sh: conpot: not found ~ $

PWNZ1337cr3w, 2022-12-07:
Hi, 
I try run conpot but i have error:

> docker run -it -p 80:80 -p 102:102 -p 502:502 -p 161:161/udp --network=bridge honeynet/conpot:latest /bin/sh

~ $ conpot -f --template default

/bin/sh: conpot: not found

CentOS 4.18.0-348.7.1.el8_5.x86_64
Docker version 20.10.21, build baeda1f

xandfury, 2023-02-15:
The `conpot` binary should be inside the docker image. I am guessing you are trying to `conpot` from the host.

johseg, 2023-02-25:
/home/conpot/.local/bin is not in $PATH. Run
`/home/conpot/.local/bin/conpot  -f --template default`

Harshiths1, 2023-10-19:
Try running using the following command , It should solve the problem :
sudo docker run -it -p 81:81 -p 103:103 -p 503:503 -p 160:160/udp --network=bridge
honeynet/conpot