mushorg/conpot #577: IPMI server not responding

aryans90, 2022-12-29:
Conpot is running in a container and ipmi server is listening on port 6230 but it is not responding back. 
On running ipmitool command:
`ipmitool -I lanplus -H <IP_ADDRESS> -U <username> -P <password> user list`
Output:
Error: Unable to establish IPMI v2 / RMCP+ session

Error in conpot logs:
2022-12-29 09:27:32,445 New IPMI traffic from ('10.16.12.35', 55604)
Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 908, in gevent._gevent_cgreenlet.Greenlet.run
  File "/home/conpot/.local/lib/python3.10/site-packages/gevent/baseserver.py", line 34, in _handle_and_close_when_done
    return handle(*args_tuple)
  File "/home/conpot/.local/lib/python3.10/site-packages/conpot/protocols/ipmi/ipmi_server.py", line 100, in handle
    self.session = FakeSession(address[0], "", "", address[1])
  File "/home/conpot/.local/lib/python3.10/site-packages/conpot/protocols/ipmi/fakesession.py", line 55, in __init__
    self._initsession()
  File "/home/conpot/.local/lib/python3.10/site-packages/pyghmi/ipmi/private/session.py", line 623, in _initsession
    self.logoutexpiry = _monotonic_time() + self._getmaxtimeout()
  File "/home/conpot/.local/lib/python3.10/site-packages/pyghmi/ipmi/private/session.py", line 756, in _getmaxtimeout
    while incrementtime < self.maxtimeout:
AttributeError: 'FakeSession' object has no attribute 'maxtimeout'
2022-12-29T09:27:32Z <Greenlet at 0x7f34fd24c400: _handle_and_close_when_done(<bound method IpmiServer.handle of <conpot.protoco, <bound method BaseServer.do_close of <DatagramServ, (b'\x06\x00\xff\x07\x00\x00\x00\x00\x00\x00\x00\x0)> failed with AttributeError

To debug this:
`tcpdump -i any port 6230`
09:27:16.397931 eth0  In  IP 10.16.12.35.55604 > 00d841657a88.6230: UDP, length 23
09:27:17.403425 eth0  In  IP 10.16.12.35.55604 > 00d841657a88.6230: UDP, length 23
09:27:19.409200 eth0  In  IP 10.16.12.35.55604 > 00d841657a88.6230: UDP, length 23
09:27:22.416080 eth0  In  IP 10.16.12.35.55604 > 00d841657a88.6230: UDP, length 23
09:27:26.425055 eth0  In  IP 10.16.12.35.55604 > 00d841657a88.6230: UDP, length 23
09:27:27.430877 eth0  In  IP 10.16.12.35.55604 > 00d841657a88.6230: UDP, length 23
09:27:29.438140 eth0  In  IP 10.16.12.35.55604 > 00d841657a88.6230: UDP, length 23
09:27:32.445580 eth0  In  IP 10.16.12.35.55604 > 00d841657a88.6230: UDP, length 23

As it is clearly visible, packets are coming in but there are no outgoing packets.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the conpot server with default configuration.
2. Execute the ipmitool tool command as given in the line 3 of the bug description.
3. Error will be shown: Error: Unable to establish IPMI v2 / RMCP+ session

**Expected behavior**
Should return the user list specified in ipmi.xml configuration file.

**Desktop (please complete the following information):**
 - OS:  Ubuntu
 - Version 22
 - Container with base image ubuntu

M4rdc0re, 2024-07-17:
Hi,

Did you solve it?

AzazelSensei, 2026-08-28:
Opened #624. FakeSession skipped Session.__init__, so newer pyghmi died on maxtimeout inside _initsession and the UDP packet got no reply.

closed 2026-08-30 by glaslos