mushorg/conpot #581: Omit certain IP Address from logs?

KKAB-FK, 2023-02-07:
Hi,

I have a python script that "will" be running on the same host as Conpot and the script writes values to some Modbus registers!
As a consequence this "register writes" are being logged and displayed as an "attack log" and being sent to my syslog server!

Is there a possibility to omit a certain IP from being logged?

Br
Felix

glaslos, 2023-02-08:
Unfortunately there is no feature for this, but it should be easy to add if you want to give it a try.

KKAB-FK, 2023-02-09:
Could you give me a hint on where i should look?
Ive looked through all log related code and could not find the right place in the files in order to do my change!

Br
Felix

glaslos, 2023-03-29:
[Here](https://github.com/mushorg/conpot/blob/master/conpot/core/log_worker.py#L110) when you get the event, check the source IP address and then skip the log processing.

Neoshka1337, 2023-03-29:
> [Here](https://github.com/mushorg/conpot/blob/master/conpot/core/log_worker.py#L110) when you get the event, check the source IP address and then skip the log processing.

Thanks, i'll try :)

glaslos, 2024-02-27:
@Neoshka1337 have you tried?

Neoshka1337, 2024-02-29:
@glaslos Unfortunately no, I was already running out of time to finish my thesis and didn't have time to test this feature :(