mushorg/conpot #585: conpot run error

ringringboy, 2023-09-11:
```
Ubuntu18.04 + conpot0.6 + python3.6 + virtualenv
```

```
(conpot-env) syr@syr:~/desktop/conpot-env$ conpot --template default
/home/syr/desktopconpot-env/lib/python3.6/site-packages/pyghmi/ipmi/private/session.py:29: CryptographyDeprecationWarning: Python 3.6 is no longer supported by the Python core team. Therefore, support for it is deprecated in cryptography. The next release of cryptography will remove support for Python 3.6.
  from cryptography.hazmat.backends import default_backend
Traceback (most recent call last):
  File "/home/syr/desktop/conpot-env/bin/conpot", line 45, in <module>
    from conpot.protocols.IEC104.IEC104_server import IEC104Server
  File "/home/syr/desktop/conpot-env/lib/python3.6/site-packages/conpot/protocols/IEC104/IEC104_server.py", line 17, in <module>
    from conpot.protocols.IEC104.DeviceDataController import DeviceDataController
  File "/home/syr/desktop/conpot-env/lib/python3.6/site-packages/conpot/protocols/IEC104/DeviceDataController.py", line 19, in <module>
    from conpot.protocols.IEC104.frames import *
  File "/home/syr/desktop/conpot-env/lib/python3.6/site-packages/conpot/protocols/IEC104/frames.py", line 258, in <module>
    class QOS(Packet):
  File "/home/syr/desktop/conpot-env/lib/python3.6/site-packages/scapy/base_classes.py", line 345, in __new__
    for f in dct["fields_desc"]
  File "/home/syr/desktop/conpot-env/lib/python3.6/site-packages/scapy/base_classes.py", line 345, in <listcomp>
    for f in dct["fields_desc"]
  File "/usr/lib/python3.6/inspect.py", line 2488, in __init__
    raise ValueError('{!r} is not a valid parameter name'.format(name))
ValueError: 'S/E' is not a valid parameter name
```
I only use
```
pip3 install conpot
conpot --template default
```

closed 2023-09-11 by ringringboy

HantaoGG, 2024-03-18:
@ringringboy  Pls how did you solve this problem, I had the same problem Thank you very much

HantaoGG, 2024-03-18:
@ringringboy  I also wanted to ask you if your MAC address modification feature is available, it's at the very bottom of test.cfg

ringringboy, 2024-04-10:
> @ringringboy I would also like to ask you if your MAC address modification feature is available?

Sorry for not visiting git-hub for a long time, I just saw your reply today!
(1) I used the command "conpot -f -t default" and it worked! 
(2) I scanned my conpot with Nmap and it showed the MAC address, but I didn't resolve it.

danperrout, 2024-10-11:
I found out that "conpot" doesn't work with the new version of "scapy", so you should use version 2.4.5 of it.

After installing conpot, you must downgrade your scapy using

`pip install scapy==2.4.5`