mushorg/conpot #592: Python 3.11 breaks conpot

t3chn0m4g3, 2024-02-27:
Python 3.11 breaks conpot, reason is the cpppo library which is not actively maintained.

For conpot to work with Python 3.11:
- Remove `cpppo` from `requirements.txt`
- Fork `cpppo` from [here](https://github.com/crmaxj/cpppo/tree/maxj/fix_python_3_11_CodeType)
- Merge this [PR](https://github.com/crmaxj/cpppo/tree/maxj/fix_python_3_11_CodeType) into the fork
- Setup `cpppo`
- Setup `conpot`

ethan-thomason, 2026-03-28:
@glaslos — happy to help with this. I can confirm both failure modes independently and have some initial thoughts on approach. Will be picking this up in a few days but documenting findings here now so I don't lose the thread.

---

**Failure 1 — `fs` / `pkg_resources.declare_namespace()` crash on Python 3.12**

This hits before cpppo even enters the picture. Python 3.12 removed `pkg_resources.declare_namespace()` as part of the broader importlib migration, and the `fs` package (PyFilesystem2) calls it in its `__init__.py`, causing an immediate `ModuleNotFoundError` at conpot startup.

Ideas for fix:
- Check whether a newer release of `fs` has already resolved this — if conpot is pinned to an old version, a simple version bump may be enough
- If not, audit the actual conpot usage of `fs` (likely `grep -r "import fs\|from fs" --include="*.py"`) to understand scope. My guess is it's used in a few places for path/file handling that could be replaced with `pathlib` / standard `os` calls without much pain
- Worst case, a minimal shim or explicit importlib namespace registration could unblock it without touching the rest of the codebase

This feels like the smaller of the two problems and probably the right thing to fix first since it blocks even getting to the cpppo error.

---

**Failure 2 — `cpppo` incompatibility on Python 3.11+**

`cpppo` is unmaintained and breaks on 3.11+ due to syntax/API changes that have accumulated. The approach @t3chn0m4g3  outlined (fork cpppo, merge the community fix PR into the fork, depend on the fork) technically works but trades one problem for another — now conpot depends on an unofficial fork of an unmaintained library, which creates a long-term maintenance burden and makes the dependency chain harder to audit.

Ideas for fix:
- **Option A (fork + patch):** Follow t3chn0m4g3's approach as a short-term unblock. Lowest effort, highest risk of recurring rot.
- **Option B (replace with maintained Python library):** Evaluate whether `pycomm3` or another actively maintained EtherNet/IP/CIP library can cover conpot's actual usage of cpppo. If conpot only uses a narrow slice of cpppo's API, this may be more tractable than it looks.
- **Option C (Apache PLC4X):** I have a contact at https://github.com/apache/plc4x, which has Python bindings and supports EtherNet/IP among many other industrial protocols. Apache-backed, actively maintained, multi-protocol — a more serious long-term foundation than any Python-native alternative I'm aware of. Worth a conversation about whether cpppo's role in conpot could be replaced by PLC4X bindings.
- **Option D (conditional import / graceful degradation):** If EtherNet/IP support is optional in conpot's architecture, gate the import so startup doesn't fail hard when cpppo is broken — lets users run other protocol templates while this gets resolved properly. Could be a quick standalone PR as an interim unblock regardless of which long-term fix wins.

My instinct is to look at Options B and C before committing to the fork path, but that depends on how deep cpppo is woven in.

---

**Game plan:**
1. Audit `fs` usage across the codebase — if it's shallow, fix that first as a standalone PR
2. Audit cpppo usage to understand whether a library swap is realistic or whether the fork path is the pragmatic call
3. Open a tracking issue (or use this one) before writing code to align on approach

Will report back with findings once I've had a chance to dig in properly. Happy to split into two separate PRs since these are independent failure modes.

ethan-thomason, 2026-03-28:
Glad to see this moving! I'll pull the branch and run it against my setup to confirm the fs / pkg_resources issue is resolved as well. Will report back.