mushorg/conpot #593: The MAC address modification permission is insufficient

HantaoGG, 2024-03-10:
When building from the current dockefile, the ability to change the MAC address will prompt 
'Changing mac address require sudo permissions. Skipping',
 I'd like to know how to make the permissions sufficient, and I'd like to try this feature, even though the operating system you mentioned is restricted.
 Thank you very much.

glaslos, 2024-03-10:
@HantaoGG we removed this feature in the latest version of Conpot. Is this an important feature for you?

HantaoGG, 2024-03-11:
@glaslos Thanks for the reply!I do want to use this feature, so where can I use it?

HantaoGG, 2024-03-11:
I'm using the latest version of the repo

HantaoGG, 2024-03-18:
Sorry to bother you again, but I'd like to know how do I make the MAC address change function work in a conpot 0.6.0 docker environment? It always says Changing mac address require sudo permissions. Skipping
 It is an important feature for me